polyend/TrackerBetaTesting #811: Choke Groups/Mute Groups

vielfrassss, 2021-02-15:
Basically, I just got the Polyend Tracker, and I am basing my use of it mostly around the sampler.
Sampling notes and sounds into it, and then using it as a rompler/mellotron/drum machine.

But one thing that really, really bugs me, that might even get me to find another product for this application is that I cant setup choke groups for the open and closed highhats when using it as a drum machine. Basically what i want is to have the open HH on channel 1 and the closed HH on channel 2, and whenever I trigger a note on channel 2 it should send a note off to channel 1 and vice versa.

I really, really like the computer menu based workflow of the Tracker with its well thought through shortcuts and its nice buttons and the BIG screen,
but this tiny little thing with the choke groups could be a dealbreaker for me.

My question, is this something that could be implemented in a coming update?
Or do you know any hack, or trick, or workaround for how this could be done?

I think if the tracker had this functionality, there are a lot of people I know who are looking for an easy to sample into drummachine, with lots of useful parameters like envelopes and compression and other effects that can quickly applied to the samples, and this could maybe fit in a yet unoccupied niche in the market. If such a feature was considered to be added, I think marketing the Trackers use as a drummachine might be quite an effective strategy.
Thats just my opinion.

aheger, 2021-02-15:
Maybe I fail to understand your problem, but wouldn't this be solved by putting closed HH and open HH on the same track?

vielfrassss, 2021-02-15:
it would solve it if i was step sequencing everything, but i want to live record the drums from an external controller that is also velocity sensitive. so i assign one "instrument"to each track, and have each track be controlled by a different pad of the controller.

is there a better way to live record drums from an external controller?

vielfrassss, 2021-02-15:
I want to live record each drum track individually, track by track and lay down the open HH first and then record the closed HH on channel 2 and have them automatically mute the open ones on channel 1. that way i can have microtiming in the drumloop that adds a certain human feel to it.

aheger, 2021-02-15:
So basically you want to create an offset between closed and open HH that is smaller than one step? 
And if you recorded to the same track, open and closed HH would land on the same step.

tfspsound, 2021-02-15:
If you record your HH track then when you what to add your OH you could select the HH track again and hit play, hit record when you want an OH and replace the instrument. Alternatively use beat slices and just play the OH into the pattern. 
Also you could just make your HH + OH track in non live recording mode and add microtiming after, might seem like cheating but it gets the same result.

vielfrassss, 2021-02-15:
I see, i am still in the process of warpping my head around the uniqueness of the tracker sequencer.
So there are ways to do it, but you have to think a bit around the corner and work around the compromises.
If multitimbrality was added, that feature would make a lot more sense.
well here we are in the would have and might have beens.

i am curious what the devs think of live drum sequencing, if they are interested and if yes by how much.

talktogreg, 2021-02-16:
Lay down CH on Track 1 as per normal.
Then lay down OH on Track 2 with your timings. 

Then move the notes (will take micro timings with them) from T2 to T1.

You'd need to do this for each OH separately - but just how many OH do you need? LOL

Ps: move is quite easy but a bit Undocumented.
- Copy+Delete (copies to buffer and clears step)
- Shift+Copy (pastes buffer)

ThunderingMantis, 2021-03-09:
In this instance I think you are wanting to the Tracker to behave more like an MPC than, well, a tracker.

An advantage of a tracker is you can have different samples on the same track.

As others have said, you can live record the closed hi-hat on one track, then live record the open hi-hat hit-hat on the same track. Or as one person said, you could record both samples on different tracks, and copy and paste one track to the other.

The way I would do it, to get the most natural feel, is to live record, but have the instrument lane selected, so that you can play both the closed hi-hat and open hi-hat, in one take and on one track.

Where you might run into a problem is if, dye to the micro-timings, the closed hi-hat and the open-hat are recorded on the same step. One would overwrite the other. Should this happen, I would suggest you would benefit from doubling the BPM of your project, and doubling the number of steps in your patterns, not just as a workaround for this but because it sounds as though the type of music you are working on requires a high level of precision and detailed edits

DonnaLisa, 2021-03-09:
When I was using the unit i discovered several problems. One was how the note handling behavior did its thing when playing polyphonicly with an external keyboard. When I played a new note it mutes the previous note, and only playes multiple keys simultaneusly if i press for example 3 keys at the same time, but then when i play a different tritone it mutes the previous tritone instead of letting it ring out. Since it is advertised as an 8 voice polysynth as well i expect that you should be able to set it to behave like a normal polysynth.

Another issue was that it sometimes skipped notes from my external keyboard. 

And the last problem was that i was not able to adjust by how much the incoming velocity is modulating the amplitude of the note, wich resulted in some noted being super loud and some notes being super quiet, when playing dynamically on an external keyboard.

and lastly i thought that the ability to play sample slices across multiple keys is very nice, but it would be even better if you could space it out more and assign a sample slice to an entire keygroup and maybe even different velocity range, effectively allowing for some crude multisampling.
(And of course if it had an autosampler that could sample and assign those keygroups and velocity layers automatically that would be quite amazing)

You see I am trying to use the Polyend Tracker in ways that propably arent what the developers had in mind when they designes it.

For now i have returned it and got a Maschine+ instead, but i am still keeping my eye on the tracker, and if some of these features + maybe multitimbrality get added in the future i will definitely give it another shot. Because I really like the workflow and simplicity with the clicky buttons, the big matte screen, the computer based menu system and the big jogwheel. I also really like the formfactor and dimensions for this kind of deivice. And of course if you want to step sequence or edit a sequence afterwards, the sequencer is brilliant for that, and there is nothing on the market that comes close.

vielfrasssss, 2021-03-18:
Actually, I ended up not liking the Maschine+, and returned it.
The Maschine+ is great for playing presets sounds and not so good for creating sounds from scratch,
at wich the tracker is really good. And I am into making my own sounds, I cant stand presets.

And I bought the Tracker again, and I found out that you can actually adjust velocity, and that you can create kind of a round robin mode by putting multible similar one shot samples into one longer sample, slicing it, and having the slice choosen randomly. That is very useful for drum stuff.

However I discovered that I like to do this kind of fingerdrumming:
https://youtu.be/vlk9s5gPgXY?t=329
At the moment I use the 1010music blackbox for it, but if the following features would be added to the tracker, I would also be able to do it on the tracker. (with an external contoller of course)
1.Mapping each instrument to a different key/pad on the keyboard/pad controller.
2.Setting choke groups between 2 instruments.
3.Being able to map each instrument to a fixed track, and it will only play in that track.
That would solve the problem of samples being played in real time muting each other when they shouldnt. If each instrument has its own track that wont happen + it would be easier to go from recording into the sequencer to the live perform mode where you would add ratchets and temporary midi and audio effects to individual tracks while playing back the sequence.

closed 2021-12-02 by Jacknametrouble