polyend/TrackerBetaTesting #814: [1.3.0] WT Smoother doesn't work on 64 x 256 wavetables.

voltagecontrolled, 2021-02-16:
Hello,

I am attempting to use the WT Smoother effect on wavetables gathered from other sources such as https://waveeditonline.com/. These are sets of 64 waveforms that are 256 samples each. (16,384 samples in total length)

Steps to reproduce: 

- Download a sample from https://waveeditonline.com (tried at least 10-15 of them with same results)
- Load to a project via Sample Loader and select as current instrument
- Via Sample Playback, set Play Mode to Wavetable and set Window to 256
- Select Position and play a note while adjusting the Position
- Tracker will move through all 64 waves in the wavetable without interpolation / smoothing
- Via Sample Editor, select "WT Smoother" effect
- Set Window Num to 64 and Window Len to 256 and press Apply
- Go back to Sample Playback and scan through the positions while holding a note
- Tracker will now only move through 63 positions and there will be no smoothing but visible alterations to the waveform

It appears all that is happening is the wave is being shortened but not smoothed at all.There appears to be an attempt to fade to zero crossing between the sections of the sample but it is not successful and it seems like the margins between the waveforms do not line up properly.

Thanks for taking a look!

MortalJS, 2022-02-22:
Hi @voltagecontrolled 
Can you check if the problem occurs in firmware 1.5?

voltagecontrolled, 2022-02-22:
> Can you check if the problem occurs in firmware 1.5?

Sadly I cannot because I do not currently own a Tracker. If I can get my hands on one though soon I will test and report back.

closed 2022-07-14 by MortalJS