polyend/TrackerBetaTesting #815: Beat Slice - Functionality

3zana, 2021-02-16:
**Using pads to select slices.**
When in the sample playback window: Beat Slicing mode, it would be great if the pads would actually select the slice. That way one can adjust the slices more intuitively, without the extra step of pressing the "slice" button and scroll to the right slice. It would also make it more intuitive to add slices.

**Auto Slice Sensitivity**
It would be great if the auto slice sensitivity could be adjusted. Perhaps while holding down the "adjust slice"-button. Alternatively, Auto Slice could also respond to the tempo by slicing per beat.

**Doubble Click**
Dubble clicking the zoom button could turn it to its default value. This function could actually be implemented on a few other places as well, like the instrument parameters.

anemlife, 2021-02-17:
I agree with every point! This would definitely enhance the workflow 

(also if you press delete when you are adjusting the zoom it goes back to the original value, same with a lot of functions. Not sure if you knew about this way!)

jefffm, 2021-02-17:
> Using pads to select slices

Big +1 to this!

Nodoubtyall, 2021-02-20:
Yes! Especially to using pads to select slides 🤞

3zana, 2021-02-28:
**Slices Saved To Sample**
Another cool thing would be if the slice-points could be saved to the sample. As an alternative to being saved in the project. Similar to how cue-points are saved in the songs in Serato. This means that if a sample is used in another project, it is already sliced. Could be handy for drum loops.

dirtysampling, 2021-07-05:
I agree with one minor point, I think that select the slice should be Shift+Pad, so you can still audition the slice you want by hitting the pad and you can hear the end and start point of the current slice without fiddling around.