polyend/TrackerBetaTesting #816: [1.3.1beta3] Encoder not working. Broken?

swindus, 2021-02-17:
Please check this video turning my encoder on the Polyend Tracker. It's not usable at all. Is it a software or a hardware problem? What can be done here? Thanks!

https://user-images.githubusercontent.com/6263942/108219263-d70b9d00-7135-11eb-8fcb-ebfc9f237eda.mp4

swindus, 2021-02-17:
Check also this issue: https://github.com/polyend/TrackerIssues/issues/656

birdydy, 2021-02-17:
People @ Polyend will certainly get back to you but I had exactly the same encoder behaviour, and it was a hardware problem.

Jacknametrouble, 2021-02-17:
We'd like to help you with that. Please reach us here https://help.polyend.com/submit_ticket

swindus, 2021-02-18:
Ok, thanks. Looks like a faulty encoder and my unit is on the way back to Poland now.

closed 2021-02-18 by Jacknametrouble