polyend/TrackerBetaTesting #821: Controlling Games

tfspsound, 2021-02-18:
Any chance we could have an alternate ‘joypad’ set up for games mode, that is more in line with console gaming? 
[image]

tfspsound, 2021-02-21:
Just try playing Pinball, it’s a game with only two buttons and they are the wrong way round. Be nice for a joypad option somewhere in the config section at least. I know it’s not critical when there is stuff to fix etc but it would really complete the games section for me.

JR2Media, 2021-02-22:
The NES controllers didn't have diagonal movement.

tfspsound, 2021-02-24:
> The NES controllers didn't have diagonal movement.

And?

JR2Media, 2021-02-26:
Meaning NES games wasn't built to recognize diagonal movement and games don't require it. Thus implementing it isn't really required.

tfspsound, 2021-02-27:
So how does a game like cobra triangle work with its isometric viewpoint? How can I fire diagonally on contra? I made this suggestion because the pads are considerably larger than the NES joypad so if diagonal input is possible (it is) then a diagonal button would do the job that pressing up and left (eg) simultaneously on the NES would have done. 
I don know how the NES worked I don’t really care, this was just an enhancement suggestion, to which you just added a pedantic ( and quite possibly erroneous comment to. Thanks for the negative input you went out of your way to add.

JR2Media, 2021-02-27:
I do apologize if I came across negatively to you. I lived on the NES in my younger days and I would much prefer the controls to remain as close to the original as possible. But that's just my feelings and doesn't represent the feelings of others.

You must expect some people will not agree with ideas put forward and that is the spirit of the open source community (except PT isn't open source at this point). Discussion is not and should never involve a personal emotions.

tfspsound, 2021-02-28:
No worries, I appreciate your right to an opinion, I supported your move to open source the firmware on here. I think it became personal because it was the only response to an uninteresting proposal and it was negative. That and we were already playing NES games on a PT so I found it partially irrelevant. I understand, nostalgia is a powerful thing, I feel the same way about many things that we hold dear to us but sadly what the future does with them is out of our control. 
 I was only forwarding the suggestion as an alternative in the first place knowing some people will have memories of playing emulations on PCs and in that sense the PT is emulating PC gaming very well. However I just thought it would be good to be able to control it using a dpad styled control system making PT like a big gameboy. The silicon pads are too big to cover the diagonals effectively hence the suggestion for the extra control buttons. 
To truely take the emulation to the next level PT would need to be able to use a USB host, so we could connect NES mini joypads and play that way as a third option. 
All the best.

ClaidArremer, 2021-04-09:
I couldn't agree more that the current control scheme is terrible, the D-pad is on the right and the buttons are on the left! However I am hoping Polyend will allow us to rebind the controls in the future and the picture at the top of this page would be a perfect layout.