polyend/TrackerBetaTesting #823: Hold down Alice’s pad to adjust start position

cclcng, 2021-02-18:
### A feature request :

In the interest of greater fluidity when editing samples / slices , i thought of a neat idea foR editing start points.... so, holding down a slice’s  corresponding pad, would enable the user 🥈 adjust the start point uv that slice until the pad is  released. Thanks !!

Nodoubtyall, 2021-02-20:
Yes, we need this 😊

soffer, 2021-02-28:
Great idea

darenager, 2021-07-19:
+1 yes it would speed up slicing.

bambrose, 2021-07-20:
Been asked for before. But yes this would enhance workflow considerably

danapeks, 2021-07-20:
Just linking through previous requests for this as there are multiple over the last year #217 #335 #485 #782 #815 #823
Really be good to get an update from someone at Polyend as to whether this is being looked at and is possible?

Jacknametrouble, 2021-12-25:
Would be a cool addition @ambv

trutypesounds, 2021-12-25:
+1 yes pls

danapeks, 2021-12-25:
Even just using the pads (rather than holding down) to select the slice you are editing would be a great improvement

Jacknametrouble, 2022-01-15:
Implemented in v1.5.0

closed 2022-01-15 by Jacknametrouble