polyend/TrackerBetaTesting #824: [1.3.1b3] the magic of midi channel 11

somelamp, 2021-02-20:
I sadly can't say how to reproduce this yet, but I figured it was at least worth putting up an issue note and video.

So, something magical happens on midi channel 11. It only happens on channel 11. Regardless of what steps might cause it, sometimes the pads will default to chromatic, but in reverse order. (i.e. highest octave on the lowest pads, lowest octave on the highest) When this happen, it ignores scales and root note and occurs in both record and play mode.

This can be remedied simply by powering off and on, or even by opening a different project. I've been unable to reproduce this other than it happening randomly when starting a new project, which another user on the discord has confirmed. I'm going to guess this is an artifact of a previous implementation of midi or the pads.

In the video, while you sadly can't see the blofeld, I'm simply switching between the 11th and 12th midi channels on both the blofeld and the tracker.

I'd like to apologize in advance for the awful compression and clipping on the video I used my default camera app on accident and didn't feel like rerecording heh... listen at a lower volume!

https://youtu.be/duhNIAX8FXo

mFoxRU, 2021-03-24:
I have the same issue with midi channel 11. Using latest version (1.3.1 Final)

Unfortunately video was recorded rotated, but the camera was close enough so actual notes input can be seen.
https://www.youtube.com/watch?v=mwkPG0iBPXg

MortalJS, 2022-03-19:
No response. Closed.

closed 2022-03-19 by MortalJS