polyend/TrackerBetaTesting #825: [1.3] Auto folder splitting not sequential by file name

hepto, 2021-02-20:
I have some folders with over 100 files inside so the Tracker 'splits' the folder into groups of 100 as expected (though I'm not sure why this is necessary).

However, the split does not appear to be on file name, such that alphabetically they are no longer ordered. Each batch of 100 has a non-sequential selection of files.

For example I have a folder with almost 400 wavs named "vocal 001" "vocal 002" and so on.  Therefore I would expect the Tracker to put the first 100 in the first 'split' folder, 100-200 in the second, and so on.

However, instead I get a selection across the whole set in the first batch, then another selection in the next set and so on.

Other than being non-obvious, this also means that samples that are connected to each other (like different slices of a vocal) are no longer next to each other for easy selection.

talktogreg, 2021-02-21:
I can concur with this.

My guess is that the Tracker is splitting by the default file order .. which would more often than not be 'creation date'

RadicalAns, 2021-04-05:
I can also confirm this bug in 1.3.1 Final.

MortalJS, 2022-02-22:
Hi @hepto 
Can you check if the problem occurs in firmware 1.5?

hepto, 2022-02-24:
Unfortunately it still does occur in 1.5.

Here are some screenshots showing it on a couple of folders.

`Bass` folder has been split into 2:

[image]

First one lists some in order, but note the jump from `Bass 5.wav` to `Bass 7.wav`:

[image]

`Bass 6.wav` is in the second folder (101-200):

[image]

You can see the same thing with the `Drums` folder that has been split into 3.  The sequence of `Bongo ....wav` has split across the folders.

[image]
[image]
[image]