polyend/TrackerBetaTesting #826: Enhancement Request: Pattern Changing in Perform Mode

clickysteve, 2021-02-20:
Hello.

I love Perform mode. It's awesome. I have two feature requests which I believe will make it even better though

### Change all track patterns at once

I can't see any way to change all tracks at once (either shift them up and down, or to a certain pattern). The only way to do this is to hold down all the track buttons at once and move the jog wheel or arrows. That is very physically difficult to do. My request is to add a shortcut to allow global shifting.

### Faster Selection of Specific Pattern Using Pads

When switching patterns, you can only cycle through them in Perform mode. This makes it difficult if you want to jump to a specific pattern that is more than a few numbers away. It would be amazing to have the kind of selection screen you display for Notes - so that you could quickly select a specific pattern via the pads. This change could be selected with a modifier key to be either instant or to change at the end of the pattern. The latter would be preferable if only one is possible.

tfspsound, 2021-02-21:
Hey Steve. A few of us had been talking about this/ similar issue on #413. It looks like something might be happening in os 1.4 so fingers crossed on this one.

clickysteve, 2021-02-21:
> Hey Steve. A few of us had been talking about this/ similar issue on #413. It looks like something might be happening in os 1.4 so fingers crossed on this one.

Thanks @tfspsound ! That would be awesome.