polyend/TrackerBetaTesting #828: [1.3.0 and 1.3.1b3] Clipping with LP filter Resonace 100

karachinsky, 2021-02-21:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

MetabaronCZ, 2021-02-21:
I have also some problems with resonance with single cycle waveforms. Thank you for reporting this. I found that even low resonance about 15 is making some weird noises sometimes - mainly when using Filter LFO. Safe value is about 5.

Filter resonance functionality should be revised.

modulationsource, 2021-02-21:
It seams there is no headroom in the internal audio path.
So if the sample is normalized to 100 % and the resonance adds volume, it clips internally.
At the moment you can solve this by normalizing the sample to 50-75 %, than in most cases the clipping is gone.

karachinsky, 2021-02-21:
> At the moment you can solve this by normalizing the sample to 50-75 %, than in most cases the clipping is gone.

Clipping gone at 25% for the low notes, but now I get same noise playing that sample in higher octaves.

MetabaronCZ, 2021-02-21:
@modulationsource Oh, thanks for tip. I will try it.

rjvaleo, 2021-02-24:
I haven't done extensive research on the exact nature of the clipping, but I have found that there is serious clipping with resonance turned up past 50% on a sound with moderate volume.

MortalJS, 2022-02-22:
Hi @karachinsky 
Can you check if the problem occurs in firmware 1.5?

MetabaronCZ, 2022-02-24:
Hello. I just tried it and the problem is still there in 1.5.0.

As has been said, the problem is with small headroom of tracker. If I load a piano sample and play a sequence and play with filter cutoff, just setting resonance to 5 - 10 starts clipping sound and higher values start distortion. And same problem is of course with filter LFO.

MortalJS, 2022-02-28:
Still in 1.5

closed 2023-04-21 by MiroPoly