polyend/TrackerBetaTesting #829: Using PERFORM mode as SONG creator

talktogreg, 2021-02-22:
Lots of ideas are coming in regarding changes to PERFORM mode. 

One recent interesting one was direct access to selecting a Track Clip (instead of just scrolling) by making use of the silicon pads whilst Track Screen button held. This indeed would make compiling a set of Clips together (for PERFORMing) really easy ...

... so this has lead to a new IDEA!!

What if, after you've selected some Clips in PERFORM mode, you could do something like hold PERFORM+SONG, that would create a new (next empty) PATTERN and add it to the current SONG row (ie after whoever you were when you entered PERFORM mode) or at the END if sequencer isn't running.

This would allow PERFORM mode to be a SONG builder.
- make up some PATTERNS,
- go to PERFORM
- mix and match into INTRO->VERSE->CHORUS->BREAK-DOWN->CODA->?? 
- and build the SONG up as you go

AlexNorinh, 2021-02-22:
Second this/adding to. There’s an extension for Renoise called ‘live dive’ which semi-replicates abletons clip launch, when paired with a launchpad. You ‘load’ a set of patterns/tracks into the launchpad, from then you can launch any pattern from any track whilst playing back. In addition, a great feature of this extension is the ability for it to ‘record’ your session of live pattern mixes into a ‘song’. You can then go back and restructure, make unique/variations etc. Makes building tracks very quick and natural

keith-macleod, 2021-02-24:
Even just the ability to switch patterns while in perform mode would be great.

There's actually a few things about tracker play modes I don't really get. I'm not sure why switching to perform mode stops song playback. The trichotomy between song, loop and perform mode seems superfluous, the performance effects should be a layer which can live on top of both.

talktogreg, 2021-02-24:
> There's actually a few things about tracker play modes I don't really get. I'm not sure why switching to perform mode stops song playback. 

Keith, as an aside, here's why she currently leaves SONG mode in PERFORM mode - it's because in PERFORM mode you can select Track Clips that don't belong to the current (or even same) Pattern. So if you are on a slot in SONG mode linked to Pattern 9 and go to PERFORM and change Track 3's clip to one from Pattern 10, then the it breaks the current Pattern context.

It's been suggested by many to see if there are ways to reconcile this, but as it stands its by design and not a bug.

Back to my original suggestion - this is why I'm trying to find/suggest an upside to this to make use of PERFORM in another way that would help bridge the 'trichotomy' a bit.

keith-macleod, 2021-02-24:
Hi Greg, I'm aware that it's a design decision, but I don't believe that the functionalities need be mutually exclusive. When the song switches to say, pattern 4, then all tracks are switched to pattern 4.

I think it actually probably bothers me more that I can't switch patterns (outside of song mode) without disabling the track effects, which is the other unpleasant side effect of this strange modal exclusivity.

closed 2021-12-02 by Jacknametrouble