polyend/TrackerBetaTesting #835: [1.3.1b3] Sample Zerocrossing

tilllt, 2021-02-23:
No matter where you position the start, its impossible to find a point that has no "clicking" -> assumption: there is an offset that prevents to actually find a real zero crossing.

https://www.pixelscope.de/videos/watch/db192ad5-fce4-457f-8ccb-c543238f7bf4

The sample in question is "bass_classic_bunk_F.wav" from some of the default sounds.

There are related issues posted here, 
i.e. https://github.com/polyend/TrackerIssues/issues/539
but i think to get rid of the clicking is not be a feature request but a bug, which imho is really annoying.

MetabaronCZ, 2021-02-23:
Hi. Go to instrument settings and set volume envelope Attack and Release to 0.10. It can help.

tilllt, 2021-02-23:
Sure, thats possible. But theoretically speaking, it shouldn't be necessary if it was possible to select a proper zero crossing.

Just-Peter, 2021-02-23:
I experienced the same problem, it's almost impossible to find a zero-crossing spot in a sample. Hope the Polyend guys will look into this.

talktogreg, 2021-02-24:
Also happens with Single Cycle waveforms and Wavetables :(

thebradclay, 2021-03-01:
I find it difficult sometimes to get a zero crossing but  I am doing it right know with no clicking on this sample.  Use sample editor and crop.  Here is a quick video showing it with clicks at non-zero crossing and without clicks at zero crossing:

[click:noclick.zip](https://github.com/polyend/TrackerIssues/files/6063814/click.noclick.zip)

tilllt, 2021-03-01:
Ii appreciate that everyone is suggesting workarounds. But as far as I understand the matter the technique I am showing in the video should be sufficient to find a zero crossing, just by setting the start of the sample to it. I could download the sample and prepare it in my DAW as well, but I chose not to... Maybe there is something I don't get, but just selecting a correct sample start should give a playback without click. if it doesn't, there is a bug In polyend tracker.

thebradclay, 2021-03-01:
@tilllt What I am saying isn't a work around. It's how to trim samples.

tilllt, 2021-03-01:
If you want to trim it yes. If you want to leave it as it is and just set the start to a zero crossing, that seems impossible.

tilllt, 2021-03-01:
Maybe what you are trying to say is that it's just the preview which is buggy and doesn't let you select zero crossings. Maybe the actual processing correctly selects a zero crossing.

But that doesn't solve the issue because the preview should be identical to the actual processing or it kind of misses the point.

thebradclay, 2021-03-02:
@tilllt I'm not sure I understand the problem then because in you're video you are manually searching for a zero crossing. But you are talking about "preview". What I am saying is if you want to make sure you have a zero crossing then you need to open the sample in the editor and adjust it.

tilllt, 2021-03-02:
Ok, you are saying that it's possible to find a zero crossing in the _sample editor_. I am not at my tracker right now so forgive me if I use the wrong terminology, but the video I posted is not recorded in the _sample editor_, it's in the sample playback settings. There you can set the start point of the sample, so you can non destructively set the start and end of a sample.

 Regardless of how it is working in the _sample editor_ that you mentioned, it should be possible to find a start point at a zero crossing for the _sample playback setting_ 

Because maybe I don't WANT to crop the sample, maybe later I decide to move the  sample start further to the start of the sample. That's why your solution is just a workaround for a bug. Like I said before, as a workaround I could copy the sample into my DAW, edit it there and copy it back, but that's what I want to do.

Or am I completely misunderstanding you? Sample editor != Sample playback settings. Both should work.

thebradclay, 2021-03-02:
I didn't notice you mentioning before that you were concerned about the destructive nature of cropping. Having said that I am sitting in front of my tracker now with that same sample opened in the sample playback area, I have adjusted the sample and have no clicks.....I'm wondering what else we can try here. 

https://user-images.githubusercontent.com/73397631/109686994-d1a84c80-7b50-11eb-821d-1e4046adecd3.mov

tilllt, 2021-03-08:
I am not "concerned" about the destructive nature of the sample editing, i am using a FEATURE of the Polyend Tracker in the way it is supposed to be used. In any case I am slowly getting annoyed by your replies because obviously you want to suggest I am doing something wrong. I made a video of what precisely I tried to do and it is obvious that I cannot get rid of the clicking noise. If that works for you, great. Besides: the quality of your video wouldn't allow to hear the clicking noise I am talking about. Which can be reduced almost nothing by moving the start, but not _completely_. So, @thebradclay unless you have anything really constructive to say in this matter, please just refrain from commenting, ok?

thebradclay, 2021-03-08:
I'm trying to help you. 

Based on your initial post I am showing that it is possible to find a zero crossing on the sample you specified in the . Speaking on the video quality, the video is such that it allows for direct upload as there is a size limit. The audio in the video is fine. 

I used the word "concerned" because you said, "Because maybe I don't WANT to crop the sample" that shows concern for this. 

good luck with your issue.

nitelfy2020, 2022-01-05:
> Also happens with Single Cycle waveforms and Wavetables :(

Yes. And it is very annoying. Wavetable could be very very useful if that would be fixed.

nitelfy2020, 2022-01-05:
Same with the wavetable playback. It seems to offset the zero point which creates a lot of uncontrollable artifacts.

MortalJS, 2022-02-22:
Hi @tilllt 
Can you check if the problem occurs in firmware 1.5?

tilllt, 2022-02-23:
> Hi @tilllt 
> Can you check if the problem occurs in firmware 1.5?

Hopefully next week I have some time to test!

MetabaronCZ, 2022-02-24:
Helo, I think it is still there.

Polyend has a sample pack called **Polyend Tracker Wavetables & Tones pack by Michael West** (https://www.youtube.com/watch?v=dDkytKO2fh8). Load simple single cycle sine wave (file Single Cycles/Sine.wav) and you will see the end of sine wave is cropped. And the end of wave cannot be reached by scrolling and zooming.

And there is also folder with wavetables which comes with Polyend Tracker and if you load waveteble file **Samples/Wavetables_2048/wt1.wav**, you will see it also cropped in 2048 resolution.

Are the samples mangled when loaded tho tracker project?

closed 2023-04-21 by MiroPoly