polyend/TrackerBetaTesting #837: Pattern Select in Pattern Screen Latch - maybe by double tap?

wraggal, 2021-02-24:
In the pattern screen i love hot swapping between patterns by holding the screen 'pattern' button and going wild on the grid buttons.

It would be great if I could latch the pattern screen button, maybe by double tapping?, this would allow me to hot swap between patterns without holding down the screen button. It would stop the risk of accidently overwriting notes of a pattern when doing this hot-swapping, which can happen if you accidently let go of the button and press a grid square, which can happen a lot if you're also doing other things (ie controlling another device with your other hand)

thanks

tfspsound, 2021-02-24:
Agreed, swapping between patterns in the Pattern screen is almost a performance mode of its own. Adding the latch ability to this and a few other screen options (Step, Preview) would be very helpful.

icedCreamEyes, 2022-06-05:
Yes! I would love this! I find this way of performing really useful