polyend/TrackerBetaTesting #838: Faster scrolling with jogwheel (increase scroll increment button)

Nodoubtyall, 2021-02-24:
Hi 😊
I am scrolling insanely with the jog wheel as the increment is unajustable (the sample is of course zoomed out) . 

Example: I often import samples longer than 40 sec, and it takes forever (I counted 27+ rotations, to go from start to end) to scroll through it. This is too slow and I often find myself scrolling forever when setting start and end points, throughout the sample, when selecting small bits in import mode. 

THUS:
It would be a nice solution (and less strenious on the jog wheel) if I could adjust the increments, preferably by holding down a button. 

Great gear BTW!😊 But I am destroying the jogwheel quickly! 😬
Best regards

Thus

LetsSeeIfWeSolveThis, 2021-03-03:
I also notice this adjusting the length of the granular synthesizer, need something faster (shift+wheel). I think I mentioned this before.

tfspsound, 2021-03-05:
#439 possibly earlier and subsequent posts support this suggestion too. 
Maybe holding the parameter you wish to adjust quickly could accomplish this? 
Maybe setting a zoom level in Config that made zooming in and out a toggle system, maybe tiered 0, 50, 100. a single / series of presses would allow for quicker editing of samples and wave tables.

Nodoubtyall, 2021-03-12:
UPDATE: I RETURNED IT AND GOT A NEW ONE. NOW IT WORKS.

closed 2021-03-12 by Jacknametrouble

tfspsound, 2021-04-30:
So what does this mean for the rest of us? Is my jogwheel broken?