polyend/TrackerBetaTesting #840: Stereo sample on 2 synced tracks?

soffer, 2021-02-27:
Wouldn't it be great if we could use a stereo sample file - and the Tracker would actually split those to 2 mono samples on 2 synced tracks. One panned L, the other R?

LetsSeeIfWeSolveThis, 2021-03-03:
I did an entire song syncing two files (L+R) manually, this could be great.

Lislegaard, 2021-03-29:
This would be freaking amazing! I also end up basicly using two tracks with the same sound and use either micromove or position + hard panning to make a stereo sound but it would be MUCH nicer to be able to do this with just syncing the tracks to use left or right channel of a sample.

I would love it se could choose just "add as stereo sample" when importing and it could f.ex. always be on track 7/8 if that makes things easier or just have it so that it chockes the next track. So that for exsample if you input a stereo instrument on track 1 that would be the same as having that step on both track 1 and 2.

Also then it would be amazing to be able to sample stereo recordings with the sample recorder.

OmskInfo, 2021-07-16:
Yes, I would love to use stereo samples somehow without having to duplicate everything. Since there are 8 tracks in Tracker, I would happily either:

1. Have a stereo mode with only 4 tracks (each track therefore being able to hold stereo samples)
2. Individually configuring pairs of tracks as stereo. For example, I could set tracks 1-4 could be mono while 5/6 and 7/8 are stereo tracks (so it looks like there are 6 tracks in total).

Of course, having stereo playback on the 8 tracks that already exist would be amazing, but I totally understand Tracker might not have the CPU resources to do it. But with 8 tracks, Tracker obviously has the resources/voices to transform some pairs of mono tracks into stereo.

closed 2021-12-02 by Jacknametrouble

drone-rites, 2022-02-14:
I noticed this was closed without indicating the reason. Is the feature planned for a future update or is this a "will not implement"?

astrochili, 2022-05-29:
This would allow you to record stereo synths to samples and have the entire project in one box. That would be ideal to work with external instruments.

Reopen please 🙏