polyend/TrackerBetaTesting #841: [v1.3.1 beta 3] Jog wheel only works at slow speeds

pbinggeser, 2021-02-28:
### Steps to reproduce

My jog wheel works perfectly fine at very slow speeds, but normal use to quickly adjust things causes the values to instead just hop around randomly up and down around the current value I had selected, sometimes advancing by a few dozen values randomly.

https://user-images.githubusercontent.com/7341283/109407970-d9969f80-794a-11eb-9433-2c243ffb6fb9.mp4

This jumpiness at quick speeds seems to be consistent across each place the jog wheel is used, adjusting parameters, cropping samples, selecting instruments, etc. 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
This didn't use to happen so I'm concerned it might be a hardware issue somehow. I only started noticing it after the recent beta, but not immediately enough to know that was the cause. It's happening every time I use the tracker and kind of renders the jog wheel to be useless since it's no longer faster than the arrow keys.

* Does this bug happen in previous firmware versions?
Yes. I've gone back to firmware 1.2.0 that was shipped on my tracker and the issue is still there.

* Does this bug happen in every project?
Yes

doelman, 2021-02-28:
I've the same problem, really slows down the workflow. When you want to crop a sample for example it takes ages.
Ironically enough it sometimes works better when you move the joy wheel slower.

Just-Peter, 2021-02-28:
It looks like the CPU can not handle the data when speeding up of the jog wheel. Too much data at once.

pbinggeser, 2021-02-28:
That's what it seems like to me too. It's as if the sampling rate of checking the wheel position has decreased to the point that it's possible to scroll nearly 360 degrees before the next "_what position is the jog wheel in_" check is made, tricking the Tracker into thinking I scrolled slightly backwards instead of very quickly forwards.

imabigdipper, 2021-02-28:
I had a similar issue and it turned out to be hardware related. Sent in my tracker and got a replacement and haven't had the issue since. I'd send Polyend support an email and see what they think

ambv, 2021-03-01:
@pbinggeser if you bought through a distributor, write them to exchange your unit to a new one. If you bought directly, contact us through help.polyend.com to get a replacement.

closed 2021-03-01 by ambv

kressh, 2021-04-15:
Sorry for commenting on closed issue, but I experienced the same problem and thought this could be helpful if I share my solution.

On a 3rd day after my Tracker arrived jog wheel started to lag when turning counter clockwise and couple days later was completely unusable. It drove me crazy so I stopped using Tracker for a week and today finally I got some time to dig and fix this issue.

Turned out problem was because of excess lubrication. Oil leaked to the circuit and prevented wheel mechanism to work properly. To fix this we just have to wipe the contacts clean. No need to disassemble the whole unit, just the jog wheel. I took photos of the whole process.

[image]
[image]
[image]
[image]

Try to keep this thing compressed or you might lose the tactile feedback ball:

[image]

Also clean up this contacts:

[image]
[image]
[image]

bambrose, 2021-04-15:
Superb post. Thank you for taking the time to photograph and share

x2db, 2021-04-27:
Same issue here, I though it was Firmware related so I downgraded to 1.3.0 but jog behaved the same way. I think I'm out of "return to retailer" period so I will try to to fix it using your great post. Really don't want to be sending unit back and fort

Sebos1979, 2022-02-16:
I confirm same issue, cleaning method did correct the issue. Thank you !

XMPPwocky, 2022-12-14:
Yet another confirmation - the encoder behaved exactly like you'd expect it would if software just wasn't sampling it fast enough (at fast speeds, sometimes skipping steps, sometimes going backwards). Cleaning the encoder internals with a Q-tip and IPA seems to have completely resolved the issue.

However, opening the encoder up made the stem a little looser (those metal "clips" aren't spring-loaded, you have to bend them back into place, and they're probably fitting less snugly now then after opening), so the jogwheel is slightly wobblier now. Replacing the encoder entirely is probably still worth considering because of this.

robertknutzen, 2023-05-02:
Confirming this worked for me. I had way too much grease and somehow a single hair in mine. The hardest part is finding a 1.3 mm allen wrench. Super smooth now.