polyend/TrackerBetaTesting #848: Performance enhancement. CC added to Performance page

phaelam, 2021-03-02:
it might be fun to be able to have CC as an option for a performance selection.
and would be nice to be able to set specific values you would jump to in the slots below [from 0 -127].

I think you could get some pretty sweet results performing.

ketumatek, 2021-03-02:
I don t know what sort of CC they can make in performance mode, but it will be very cool and creative if there is new feature or commande CC for live performance...
Nice idea.

phaelam, 2021-03-02:
> I don t know what sort of CC they can make in performance mode, but it will be very cool and creative if there is new feature or commande CC for live performance...
> Nice idea.

sorry...my description is pretty bad. perhaps this help illustrate what I mean...
[image]

ketumatek, 2021-03-03:
Really cool idea...I like that...

ketumatek, 2021-04-21:
After using midi for several weeks to control external hardware definitively a must have possibility because at this time ...only roll and short pattern works so very difficult to use the performance mode...+ some crash some times with one of my synth ...the sound stay on like a drone sound...