polyend/TrackerBetaTesting #855: Modifier to Jump to Previous/Next Step with Filled-in Values

drowsy-net, 2021-03-04:
A huge workflow improvement would be a setting that allows the arrow keys to automatically jump over empty steps to the closest step with filled in values. 

This way, if you're on step 29 and the previous step with note information is step 5, you could jump straight to step 5 with minimal effort. This could be implemented as a button combination, or through a toggle setting on the bottom of the main screen.

tfspsound, 2021-03-05:
Great minds think alike #818

x2db, 2021-04-15:
This is a such basic thing to improve... am I missing anything ? why there is no more ppl to request this. So with 64 step pattern with only two notes on step 10 and 60 (no fill function would help here) I need to press the down arrow 60+ times to change anything on those two steps, in EXCEL I would do shift down arrow to get to the next populated value but shift-arrow is used for selection in tracker :(

dirtysampling, 2021-07-05:
Fully subscribe to this.