polyend/TrackerBetaTesting #858: What microcontroller is in the Polyend Tracker?

JR2Media, 2021-03-04:
Can anybody confirm if it's running this controller: **[180 MHz ARM® Cortex®-M4F Microcontroller](https://4donline.ihs.com/images/VipMasterIC/IC/PHGL/PHGL-S-A0003204168/PHGL-S-A0003204168-1.pdf?hkey=52A5661711E402568146F3353EA87419)**

MetabaronCZ, 2021-03-05:
Hi. Last year I saw a post on some musician forum, where someone disassembled his unit and showed picture of the guts. There was a Raspberry Pi chip. But I am not able to find the post anymore. Have anybody else seen it? I would like to know too.

JR2Media, 2021-03-05:
Here's the internals. Not sure about RPI but the Teensy 3.6 uses the same microcontroller. Their firmware uses Intex Hex. I've written a decoder for it that I'll publish shortly.

[image]

[image]