polyend/TrackerBetaTesting #860: Suggestion --> More controls over SD card files

anemlife, 2021-03-05:
Would be nice to get a few more controls for the samples and folders in the SD card!

For example: 

- copy/past/move samples/folders through different place
- create folder
- rename files/folders
- rename all samples from a folder (+ number them)
- apply FXs to internal samples (chorus, flanger, stretch, etc...) without having to load them to a song

(also maybe screenshot + rename + image reader? As the PT can display NES games it could maybe display images in certain formats?)

Not sure if this is even possible + how it could be implemented but it would be really great to clean the sd and to create our own custom banks!

Folkhogskola, 2021-03-12:
I also think this is a great idea. I would also like the ability to have folders within projects folder so I can group different sets of projects. Folders with other projects inside don’t currently show up on tracker but show up when sd card is connected to computer. But I give a thumbs up to the ideas above. 👍🏻

NoraCodes, 2021-04-03:
I came here to suggest this as well - my personal wishlist for the Tracker's sample loader/file manager (don't much mind where these go) is to:

- Allow deleting samples from the SD card (as #342 / #582), and
- Allow renaming individual samples on the SD card (hasn't been suggested before)

Tracker is my main musical workstation, so it's a huge pain to have to go onto my PC and scroll through the list of recordings to prune and rename them. I have a workflow for this, with `aplay` in the terminal, but I do think this would to a lot to uncouple Tracker from the computer.

closed 2021-12-02 by Jacknametrouble