polyend/TrackerBetaTesting #862: [v1.3.1.beta 4] {Shift+Up} after {Shift+Home} select all tracks, not just one (video)

LetsSeeIfWeSolveThis, 2021-03-05:
1. Staying at any step of the track (but not the first).
2. Push **_SHIFT+HOME_** and then **_SHIFT+UP_**, 
3. The Tracker **_will select directly all Tracks_** and not that single track on which you are situated.

The previous behavior was **SHIFT+double UP** to select the eight tracks.

It doesn't happens every time, but it is usual, try keeping holding **SHIFT** while changing fro

https://user-images.githubusercontent.com/72809187/110148291-8ef6a680-7dd4-11eb-92cc-d28090f484bf.mp4

m **HOME** to **UP**.

This issue is present on previous version, at least since beta2.

bambrose, 2021-03-05:
Yes absolutely. There's another ticket with exactly the same issue where two of us used identified this exact issue a couple of releases back. Very annoying . Hard to work out what triggers the behaviour however.

MetabaronCZ, 2021-03-05:
Yes. This happens to me in previous beta too.

LetsSeeIfWeSolveThis, 2021-03-05:
Yes, I reported this time and **I did close the previous one**, since they couldn't reproduce it. Concerned about it (if it is forgotten) I'm creating this new one notifying is still there on Beta2, beta3 and beta4.

I still can't find a way to reproduce it the 100% of the times.

bambrose, 2021-03-05:
No me neither.

talktogreg, 2021-03-06:
I'll help to w.r.t. to previous report.

It happens only after LIVE-REC has been entered and left. Once all rows are unselected, it won't do it again (even after REC on/off) unless LIVE-REC is used, then it'll do it again.

thebradclay, 2021-03-07:
I'm using v1.3.1b3 and it happens pretty often.

closed 2021-03-11 by ambv