polyend/TrackerBetaTesting #865: [Firmware version] 1.3 b2 machine stop working...like freeze

ketumatek, 2021-03-06:
It is the 3 time my machine freeze today.
when it is happening, I can touch any bottom nothing work.
The only way to restart is put out the cable from usb power.
if I want to shoot the machine down it doesn't work...

* How often does this bug happen? (every time, often, "it only happened once")
It is the first time it arrive but 3 times today.

* Does this bug happen in previous firmware versions?
Never arrive before same on this firmware

* Does this bug happen in every project?

I don't think so ...it is a projet I save as a new projet and where I erase everything 2 or 3 times.

ketumatek, 2021-03-06:
I made another copy of the same projet change the name ...and the bug still continu...
The tracker freeze and I can do anything.
it happens again 2 or 3 times more...
The bug appears more often when I am in song mode and I copy some slot to another.

Just-Peter, 2021-03-06:
Maybe it's better to update the firmware to version 1.3.1B4 first. Than try again of you have this problem.

MortalJS, 2022-01-13:
Closed. Power unit problems.

closed 2022-01-13 by MortalJS