polyend/TrackerBetaTesting #867: Pitch/tune sample to match BPM

ThunderingMantis, 2021-03-09:
Timestretch has been a great addition to the 1.3 firmware :-)

A feature that I'd really like to see added is the ability to pitch/tune a sample so that it matches the BPM.

E.g. I have a drum break sample, and I want it to be 8 steps. Rather than timestretching it, the Tracker tunes the sample up or down, first at semitone level, then at fine-tune level, so that it matches the number of steps I've specified.

I do this in software trackers where I don't want to preserve the pitch, because the sound quality of often better this way. It's very useful for drum breaks and non-melodic samples where I don't want timestretch to degrade the sound quality.

theorize999, 2021-03-10:
I agree with this, I hate doing it manually

rjvaleo, 2021-03-10:
where would this BPM feature be added? what screen and how would it be
implemented? the BPM changes in the song setting, if that changes this
setting would need to be tempo-locked, right, like the synched delay, and
even that is simply a delay...

this may be a good idea, but not sure how useful to many people. I'm
old-school, i'd just say do the work manually. i think what would help are
USER DEFINED MACROS so that often used combos could be run easily. No idea
how easy/hard that one is, but it could help with a lot of issues possibly.
Like this one where it's a matter of automatically doing something that is
possible, albeit with some work.
--

TWITTER: http://twitter.com/rjvaleo <http://twitter.com/#/rjvaleo>
SOUNDCLOUD : http://www.soundcloud.com/rjvaleo
TRANSELECTRONIC: http://www.transelectronic.net