polyend/TrackerBetaTesting #868: Import and play .MED files

ThunderingMantis, 2021-03-09:
The Tracker can already import MOD files, which is great, but a feature that I know would be extremely well received would be the ability to import and play .MED files.

I noticed that a lot of guys who used Octamed on the Amiga 500 and Amiga 1200 have bought the PT, or are thinking about buying it, as they consider the PT to be the spiritual successor to the Amiga. Pete Cannon called it "an Amiga in a box" :-)

There are lots of guys that still have their old floppies full of MED files, but have no way of playing them because their Amigas are broken or gone. If the PT could import MED files, it would push it over the edge to become the greatest hardware sequencer ever 😎

Thank you for considering this request, it would make many old junglists very happy, and there is still a thriving MED scene out there!

UserOverflow, 2021-03-11:
You may try use OpenMPT tracker on computer and export mods to different format which is acceptable by Tracker. I did it for my mods from Amiga (exotic, but great Digi Booster). But there are some problems:
- We have only 8ch in Tracker, but some Amiga trackers allow more - so we will lost it. Solution - edit and reduce it in OpenMPT to be more suitable for 8ch.
-  Amiga trackers (like digiBooster) have much more advanced instrument settings. Some settings will be lost. Solution: Check wht may be imported, update on computer and check and update in tracker.
- Some bugs in importing mods. I'm not sure what was the problem with importing, what from incompatibility issues, but in most cases my mods sounds much different that orginal ones. Solution: import structure, update all samples, re-edit, save, enjoy ;)

ketumatek, 2021-03-11:
I take a look to try to convert xrms files...I try with two trackers as they explain ...but never find how to import in those trackers...
I know it should be or will be possible at a moment but fur sure it will be easier if I can directly import my xrms file in the polyend...
In the same time it is not so important for me because I make some new stuff on the polyend...
But it could be a nice feature...

musojon74, 2021-03-12:
I know its' supported to bring in .mod, but sounds like the samples are downsampled on load, plus timing seems off.  And impulse tracker mods - I've not been able to load any of those correctly sadly - notes seem to come in but instruments are all white noise.  I guess it's always gonna be tough to convert lots of different instrument params, types etc to the Polyend format.  It'll never be perfect.  I have quite a lot of fun just rebuilding projects on the Polyend now :)

Breakbob, 2022-01-11:
> Thank you for considering this request, it would make many old junglists very happy, and there is still a thriving MED scene out there!
I'm with you ThunderingMantis, the ability to play mod files was one of the main reasons I bought TP myself, so I was hugely disappointed when i found out I couldn't load most of them them into PT. Claiming that PT can import .IT files is also misleading when they state on Polyend website. You literally get slots filled with white noise. I tried 100 modules and only handful were loaded properly with some missing tracks, and only ONE hardcore track by Oz-e recently uploaded on modarchive.org was playing it time, with some tracks missing of course. Converting other formats to IT format came with poor result on PT, basically also white noise. I didn't expect modules to be fully compatible with PT for the reasons musojon74 mentioned but even my Amiga Guru android app can play many modules without problems and I see missing opportunity here IMO. I only hope the developers will take a look at the feature again and make Polyend Tracker an ultimate machine for tracking and remixing old modules. For now I will have to stick with Renoise's patterns rendering and resampling them on PT. No ideal !