polyend/TrackerBetaTesting #875: Function to export each Instruments.

modmodmodmod, 2021-03-10:
Since I started using PT, all sampling work has been done with PT. Because it's much more efficient and fun than sampling to a Mac using audio interface. However, I need to mix tracks produced by PT with Mac and protools, so I may wish to perform new sampling and replace them during MIX. If the export function of PT has a function that can export an instrument, I can do a new sampling, process it, save it on the SD card, and load the created sample into protools, so I think it is convenient. I will.

talktogreg, 2021-03-10:
Agreed. 

Here's what I put up on the FB group before I knew of this GitHub (Jan 7, 2021)

#samples 
So 2 days in and I've made a discovery. You can 'import' instruments from one Project to another. Just copy the numbered instrument and matching sample file [from within source Project folder] [to destination Project folder]
Just make sure you're using a free slot (number). It's OK to rename/renumber them as long as you do both the instrument and sample to match and keep naming format.
Be careful to copy and not move, as the latter would ruin the source Project. I've successfully performed the above using the includes USB sd card reader and my Note 10 phone!! 

#Question
So the question part of this post is.. would there therefore being the chance to add a more formal Instrument Export/Import function to Tracker firmware. This would allow for creation of Libraries, sharing of Instruments and general easier curation of sounds 🙂

modmodmodmod, 2021-03-10:
Or when I load the SD card on my Mac, I think it will be the same improvement as above if the file name of each samples not be changed from PT.

talktogreg, 2021-03-10:
I think the reason they change it is to associate it with the instrument file.

As the Tracker doesn't actually change the sample (wav) when you edit it, assign effects, tune, filter etc .. that's all stored in the instrument file it creates (look in Instrument folder inside Project folder). So they've numbered them to make them a pair, ie "Instrument 1" uses "sample 1" etc.

So 'exporting' just the sample (wav) won't do what you want
(this doesn't include when you 'clip' the sample on import .. which does actually change the sample)
(not sure about when you edit the start/ends .. ??)

In any-case, I do still agree with your post - in that some form of export to share/store/library 'instruments' would be good. They've actually 'hinted' at some sort of cloud sharing service ... so maybe we're getting something soon ?!!?

phacya, 2021-03-11:
I cannot state how much I want this function.

One of the nice surprises of the Polyend Tracker, is how easy and engaging it makes sound design. I spend hours just creating sounds. It hurts significantly to not have a way to export these to use in other projects.

@talktogreg thank you for the workaround, I will be using this.

But I really really want an "Export Instrument Function" Pretty please Polyend. :)

Thanks

P

charliekemp, 2021-03-25:
This is also my biggest feature request, really hope it it gets added at some point.

I often want to import an instrument from another project, and currently have to take photos of the parameters and re-create them in the new project which is a bit of a workflow killer.

It would be great if we could copy and paste instruments between projects, as currently the copy buffer is erased when loading projects.

bambrose, 2021-03-25:
Yeah is the biggest outstanding feature not yet implemented as far as I'm concerned. Would mean I could start creating libraries of regularly used sounds

danapeks, 2021-04-19:
Agree this would be a hugely significant improvement! Yes please

Jacknametrouble, 2021-12-18:
This has been added in the newest 1.5b1 beta.

closed 2021-12-18 by Jacknametrouble