polyend/TrackerBetaTesting #878: 1.3.1b4  hard freeze when changing midi in settinfs

musojon74, 2021-03-11:
## Hi, sorry you had trouble!

### Gobto config midi. Go to notes in. Set to midi in Jack. Go to pattern. Now jump back into config. Tracker then freezes. I’ve reproduced this 3 times and also reproduced with midi cable unplugged in case my synth upset it. Froze every time and required cable pull. 

Repeated it again after choosing new project. Same result.

Jacknametrouble, 2021-03-11:
Try resetting the Config settings in the Config/Firmware.

Just-Peter, 2021-03-11:
Just tried to reproduce this on my Tracker with 1.3.1b4, no freeze at all. Tried it more than 1 one time, still no freeze.

ambv, 2021-03-11:
@musojon74 Thanks for your time reporting a freeze! Let us know after resetting Config if this is still happening. If so, please record a video so we can see your entire config and we'll try to reproduce on our end.

musojon74, 2021-03-11:
Thanks. Will try the reset and report back in a short while

musojon74, 2021-03-11:
Ok. So I found it and can reproduce after reaet. But what had happened is I changed to midi din but didn’t apply ( missed that step ) and then pressed pattern and then went back into config and it freezes. If i press apply it’s fine. Can you guys replicate like this?  So go config change notes in to midi. Then go pattern and back to config. For newbies ( like me to be fair ) that is a mistake that could occur and be quite upsetting if unsaved.

musojon74, 2021-03-11:
So here’s a google drive link to a video. Apologies for being unable to speak properly. Fried from a tough day of coding. https://drive.google.com/file/d/17FdLZY0US_HCirGSwwJNjm9Hf-6l0vFW/view?usp=drivesdk

Just-Peter, 2021-03-11:
I can reproduce this and confirm the freeze. I used both 1.3.1b4 and 1.3.1 final. I confirm this is a bug.

ambv, 2021-03-11:
Thank you for the **excellent** observation @musojon74 that this happens when an option change was neither cancelled nor confirmed. Since 1.3.1 was in pre-release on GitHub, I went ahead and updated the .ptf file on the release page with the fixed version.

If you can confirm this fixes your problem, that would be great.

closed 2021-03-11 by ambv

ambv, 2021-03-11:
that's the pre-release we're talking about: https://github.com/polyend/TrackerIssues/releases/tag/1.3.1

musojon74, 2021-03-11:
Yeah I found it :) 

Totally bug squished.  Thanks :)

DataGreed, 2021-12-19:
I have the same problem on 1.4.0 - constant freezes with midi in jack connected from blue box. When disconnected I have no such problems