polyend/TrackerBetaTesting #880: Chord creator ( resample to chords like octamed )

musojon74, 2021-03-11:
On the old Octamed we had a chord function. Not like the existing midi chord function but you would choose your sample, then choose your chord notes, and the Amiga would then make a chord on that sample slot using the current sample. Then you’d have a chord. Great feature. :-)

theorize999, 2021-03-20:
this would be especially awesome if there was an option one could use to stitch a copy on and trim accordingly for looped samples so we could have an easy way to make chords from oscillator type samples... or anything with a loop point in it. Some really nice pads could be done. I mean it's already possible with resampling... but a nice tool for that would be great. What I'm suggesting would be kind of tricky probably though but the original idea should be a snap I'm all for it.

LetsSeeIfWeSolveThis, 2021-04-14:
#974  I mention this enhancement here too to reunite all possible ideas about chords.

closed 2021-12-02 by Jacknametrouble