polyend/TrackerBetaTesting #882: [1.3.1] Some instruments gone after upgrading to 1.3.1

doelman, 2021-03-12:
Hi, 

You can maybe close this ticket directly since I have no clue on how to reproduce it, but I still want to inform the developers. After upgrading from 1.3.1 beta 3 to 1.3.1 two instruments (number 4 and 5) of the project I was working on where deleted. The notes where still there, but the instrument settings + associated samples were gone. 

This time it was quite easy to solve but I'm a bit worried about projects breaking after upgrading the tracker .. Guess I have to backup everything carefully before upgrading. 

Roeland

birdydy, 2021-03-12:
Could you recover them by downgrading to 1.3.1 beta 3 ?

doelman, 2021-03-12:
@birdydy I tried that but after downgrading to 1.3.1 beta 3 they were still gone. Added them back, upgraded back again to 1.3.1 and everything stayed ok. 
Haven't checked any other projects so far.

musojon74, 2021-03-12:
Were the files still on the sd card, just removed from the project?

doelman, 2021-03-12:
Also gone from the project > samples folder.. Very weird

doelman, 2021-03-12:
Instruments folder I mean

talktogreg, 2021-03-13:
Had you saved the Project since adding 4 and 5 ,or was it your current working Project (ie the one that auto-loads on power on) ?

If the latter, then maybe the upgrade removed the working_project which may've been holding you 4 and 5 instruments that hadn't yet been written back to saved Project folder ??

doelman, 2021-03-14:
@talktogreg i did specifically save it before upgrading. I also had some previous (save as) versions of the projects which still contained the 4 and 5 instruments. Very weird.

MortalJS, 2022-01-13:
Close. Cannot reproduce.

closed 2022-01-13 by MortalJS