polyend/TrackerBetaTesting #884: [1.3.1] External Midi Controller Issues

vielfrasssss, 2021-03-12:
I connected an external midi keyboard to the midi din port on the tracker.
I programmed a pad sound with a looping sample and a long release time.
I tried to play the sample polyphonically with chords, legato and just one note after the other.

There were 3 isses with that, 1 technical issue, and 2 usability issues.

Technical issue: Sometimes when playing one tritone and then another tritone quickly after, one of the notes of the new tritone did not sound. It was simply swallowed/skipped.
This happened with 2 known good midi keyboards.

Usability issue 1: Incoming velocity always affects the volume of the note. Velocity 1 corresponds to 1% Volume and Velocity 127 corresponds to 100% Volume. there is no way to turn this off, or change the scaling. (For example I would at least like to set Velocity 1 to correspond with 70% volume, and if possible also change the velocity curve.)
This makes it almost impossible to play the internal synth with an external keyboard, that does not allow you to set the velocity to a fixed value.

Usability issue 2: In case I want to play a pad sound, I need a different not stealing behavier, wich allows the notes with long release to ring out, even when playing a new tritone, while the old one is still sounding. The only current note handling algorithm tries to immediately replace old notes, wich is great for saving voices, but makes it impossible to play it like a normal polysynth.
My suggestion is to add the ability to select between 2 different note stealing behaviours per instrument and also allow the user to set the max voice count for an instrument when plying polyphonically + an option for legato mode would also be nice.

MortalJS, 2022-02-22:
Hi @vielfrasssss 
Can you check if the problem occurs in firmware 1.5?

vielfrassss, 2022-02-22:
Hi MortalJS, I no longer own the Tracker, however I am still following its development and I am considering getting one again in the future because I love interacting with it. The problem I was describing was the reason I returned it. It wasn't some crazy combination that is hard to reproduce. All it takes is a standard midi keyboard. If you think this is fixed I might order one from my local retailer and try it out again. Please let me know!

MortalJS, 2022-02-23:
Hi,

Today I receive some answer in request connected with Midi Controller. I'm
not sure i'ts the some problem or not. But in nearly future i'll try to
test this.
[1.3.1] unable to input notes via midi controller while pattern is playing
#904

vielfrassss, 2022-02-23:
I looked at it and #904 looks to be a different problem.

reubenfinger, 2022-02-24:
I can confirm this bug is still present in 1.5 second note played by external midi keyboard skips or doesn't sound, and persists even after a reconfig, I am also experiencing the problem of issue 904 where external input notes won't record if you press record while the pattern is playing

closed 2023-04-21 by MiroPoly