polyend/TrackerBetaTesting #886: 1.3.1 Bitrate of exported Songs 44.117 ?

weide72, 2021-03-13:
## Hi, the bitrate of exported songs is 44.117 . It should be 44.100 to avoid resampling.

### Steps to reproduce

Go to File Menu, press Export and Song

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once") Every time

* Does this bug happen in previous firmware versions? Yes

* Does this bug happen in every project? yes

ambv, 2021-03-14:
This is not a bug, 44117 is the internal sampling frequency of the device which comes from dividing the CPU frequency. Changing it to 44100 during offline export would mean that either tempo would be out of sync compared to realtime playback, or we would have to do resampling inside the device. We are producing files at the exact frequency so you can edit them as needed in a DAW.

closed 2021-03-14 by ambv

acidf2018, 2021-03-27:
@ambv  Thank you for your reply.

Is it preferable to record the samples in 44117 before importing them into the tracker ?

acidf2018, 2021-04-01:
@ambv 
It should be pointed out in the manual...

_Rendered stems are PCM .wav audio files with a sampling rate of 44100Hz/16bit._

modulationsource, 2021-07-11:
It would be interesting, what that means to the internal sample handling.
Does the sample playback happen with resampling or without ? 
Without resampling all playback would have a higher pitch and tempo by the factor of 1,00038.
Not that much, but with a 10s sample it would be 3,8 milli seconds. 
The pitch would be 0,5 cent higher, this should not be audible.