polyend/TrackerBetaTesting #887: [1.3.1] high pitched noise

WugWugs, 2021-03-13:
After updating I immediately noticed high pitched noise coming from inside the device, mostly from around the centre of it, bottom screen. No change with or without line out, or when play is on. Different cables, different ports, still the sound. So I did revert to previous versions and found out the noise comes with the final 1.3.1. and 1.3.1b4. 

1.3.0. and 1.3.1b3 do not have the issue.

It is not loud, but with nothing playing around I can clearly hear it, I guess it wont bother when doing music, but a bit afraid it can be damaging to the device.

CEEESSLAB, 2021-03-13:
I too have noticed a high pitched tone emanating from the USB-C connector area - possibly the DC switching circuitry.

talktogreg, 2021-03-14:
Just a thought, but since there is 2 different hardware revisions (original square/flat buttons, vs newer rounded/higher buttons) could you list which you have for the devs ?

They do have slightly different circuitry in them (as well as the cosmetic difference above), so it would be best to know if this issue is only affecting one revision or the other.

For the record, I have the original, and there is no high pitched noise coming from the hardware itself, but a little though the headphone.

Cheers

Jacknametrouble, 2021-03-15:
Hey, please go to Master Mixer 3/3 and turn off the Live input monitoring, this should be it.

notaprogrammer9, 2021-03-17:
I think I have the same issue with 1.30. (See my ticket on crickets)

closed 2021-12-02 by Jacknametrouble