polyend/TrackerBetaTesting #888: 1.3.1 freezing

MicahMoses, 2021-03-14:
I am having several issues after updating
Its getting an external clock from a Analog Keys as well as a keyboard auto channel

1. Keyboard input doesn't work at all
2. After a few passes of editing, it will not allow me to access the very first note line. Only allows me to start making notes at the second
3. Is finiky about starting and stoping, and eventually completely freezes transport playback (but will allow to view other pages)
4. Lots of hanging notes

Jacknametrouble, 2021-03-14:
Can you please try performing the config reset in the Config/firmware and check if it helps?

ambv, 2021-03-14:
Do the issues happen without external MIDI clock with the MIDI cable disconnected?

ambv, 2021-03-14:
I can't make heads or tails from your report so far @MicahMoses. Ideally record two minutes of video with your phone to show us how you're using it:
- what are your MIDI settings both on the Tracker end and Analog Keys
- are the hanging notes on a MIDI instrument or internal samples
- "keyboard input doesn't work" -- which keyboard?

MicahMoses, 2021-03-14:
Ok i went back to 1.3.0 and then reinstalled.
They keyboard input now works correctly
I've sent a video showing the bug.
Here is a brief explanation of my setup and the conditions.

Admittedly i have a pretty crazy midi setup.
The analog keys is the master clock as well an auto channel (channel 15 set
the same in the tracker)
They are routed together via an iconnectivity MioXM , along with various
other gear.
However to rule stuff out the issues happen when also connected directly
via din midi
(Tested directly with analog keys, digitone, octatrack, fh-2, and a
hydrasynth). Also with the trackers internal clock.
The video is with the internal clock and midi sent out to an 0 -coast and a
moog model d.

When i scroll through notes on the jogwheel, it sometimes (I was able to
catch it after the fact in the video)
It causes hanging note but also a freeze and I am am unable to navigate the
screen at all.
Pressing play a couple of times seems to fix it.
Again this is with the internal clock.
I wasn't able to replicate it this morning but last night I was still
having the issue or sometimes It will not let me go to the 1st note in the
tracker.
It happens kind of randomly and I wasn't able to get a video but I will
take a video next time it happens.

Fortunately after reinstalling the firmware the freezing issue isn't
happening as much.
Right after I installed the new firmware it was happening almost every
minute or so.
Now I have to play with it for like 30 minutes for a bug to pop up.

In the past few firmware i had seen these happen every so often but not
enough to really annoy me. Over all i am still really happy with the machine

https://drive.google.com/file/d/1SuO2u8WpOv032dYmROS6UdineUlO6hEp/view?usp=drivesdk