polyend/TrackerBetaTesting #889: [1.3.1[Firmware version] Bug title

doelman, 2021-03-14:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

doelman, 2021-03-14:
btw, resetted the config as suggested in another topic.

sunguild, 2021-03-16:
I was just coming here to post this one. I had an LFO going to cutoff, it seems to reset the phase of the LFO at about the rate of a 16th note when an external clock is received. However, when I tried adding an LFO to volume or panning to trouble shoot, those were unaffected. Have you tried this?

doelman, 2021-03-16:
@sunguild volume also effected but maybe it depends on the sample and/or LFO rate. Very frustrating, everything I think I have found the right setup to jam some new troubles pop up.

ROM72, 2021-06-24:
Same thing here, the LFO is not doing what it's supposed to do when the tracker is slaved, is super annoying.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS