polyend/TrackerBetaTesting #891: [1.3.1] Rendering freezes device

chris-roerig, 2021-03-15:
### Steps to reproduce

Select all tracks and then render to sample.

### Other important information

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? no

* Does this bug happen in every project? yes

## Notes

The screen would load the rendering image bug the progress bar would stop around 1%.
Turning off the device is the only way to quit. 
Rendering with the same name gets an error "file with that name already exists" but the file is not visible in the folder browser

dirtysampling, 2021-07-21:
Same with, rendering any track freezes my unit. Can't even turn it off. I need to unplug the cord.

coreywonggithub, 2021-09-13:
Same here. Rendering anything freezes the tracker completely. On 1.4.0

MortalJS, 2022-03-19:
No response. Closed.

closed 2022-03-19 by MortalJS