polyend/TrackerBetaTesting #895: [1.3.1] Loop Pattern not continuing with song

stuey99, 2021-03-16:
Hi

When in song mode 'loop pattern' loops the current pattern fine but when 'continue song' is pressed the song jumps back to the top of the pattern whilst still playing the current pattern.  The second pattern then follows regardless of where you were in the song.

Video of bug happening

[https://1drv.ms/u/s!AudP6qQDKfvAgd0EvliKPzrY-ZAViw?e=51I4E9](url)

Bug happens every time.   
Tracker is new and I upgraded to 1.3.1 so not sure if older version had this issue
Bug happens on every project so far

Many thanks

Stuart

stuey99, 2021-03-16:
This is 1.3.1 issue.  I changed the firmware to 1.3.0 and the issue didn't occur.  Hope this helps

modmodmodmod, 2021-03-17:
+1. I was just trying to post the same issue. The contents are as follows.

When looping slot A in song mode, there are cases I need to loop slot B without stopping the loop playback. In this case, if the continue song button is pressed once, slot B is selected, and the loop pattern button is pressed, slot A will be loop-played again.

To solve this problem, if I select slot B while slot A is playing in a loop and press the continue song button, slot B will appear to be playing in a loop on the screen, but it is actually The audible sound and the locator at the top of the song mode screen indicate that slot A is playing.

talktogreg, 2021-03-17:
Same here. Can confirm this behaviour for both 1.3.1 and 1.30

ambv, 2021-04-26:
Fixed in 1.4.0b1.

closed 2021-04-26 by ambv