polyend/TrackerBetaTesting #896: [1.3.1] Bug report - beat slice play in pattern mode = to problem

ketumatek, 2021-03-16:
I think it is like a bug or you did it like and in that case it will be a request.

I slice 2 sample with slice beat..

First bug is that the joystick go very slowly and not accurate when I want to move a slice quickly to the left..
When I do it slowly it is ok so for a small move no problem but for a big move it happen...

Second bug but I don t really know if it is one...
I can play my sample and all the part of it in pattern mode if there is nothing.
Since I rec in a track a slice...when I m playing the slice I wrote it play .
but if I want to play another part of the slice after the first one it doesn't work.
Like in live situations if a have slice but pla some other or more, when I touch the pad no sounds going out.
I have to stop and play and start playing again and at this time it works until a sample playing...

Not really easy to explain well when it is not your language...hope you understand what I mean.

To resum I can playing in slice mode if one of the slice I m using has been play in the sequencer

MortalJS, 2022-02-22:
Hi @ketumatek 
Can you check if the problem occurs in firmware 1.5?

ketumatek, 2022-02-23:
I don't think I have this problem with the last update 1.5

closed 2022-02-23 by MortalJS