polyend/TrackerBetaTesting #897: [1.3.1 and older] Tracker updating the firmware by itself

anemlife, 2021-03-16:
I've seen a couple people having this issue since at least version 1.2 I think and I'm a bit worried as this bug happens completely randomly and makes the PT still not fully reliable for live!

I know there are other fatal bugs but this one is especially annoying as you have to wait the update time + upgrade again to the firmware you were using before the bug! Also depending on the firmware it decides to upgrade, it kills your config!

Example: I was using 1.3.1 / The tracker downgraded to 1.1.17 / When going back to 1.3.1 I had to reload my whole performance values + hexadecimal config + midi delay + sidechain, etc... (all the new stuff)

The only workaround I see is to keep only the current firmware you are using on the SD folder, so the PT can't downgrade to an older one but still, I like to keep older firmwares on the SD with all the beta stuff you add and I think this bug shouldn't happen!

Also to mention I own a first and second batch and the bug happened a couple times with both versions. (in about 3-4 months)

Thanks

talktogreg, 2021-03-17:
A little Q:
- Do you hold power-button (when turning off) and wait for shutdown, or simply remove 5v usb source? Just asking as I'm wondering if for some reason, Tracker is triggering it's "load random firmware from card" after reset as noted in user manual PDF

Also, maybe best to use a 2nd SD card to hold/perform firmware updates and only keep (whichever release you're comfortable with) on performance card.

Jacknametrouble, 2021-03-17:
@anemlife this is not a firmware issue but a hardware-related thing. Reach out to https://help.polyend.com/submit_ticket so we can help you with solving this.

closed 2021-03-17 by Jacknametrouble