polyend/TrackerBetaTesting #900: A few simple to implement feature requests

robfenn, 2021-03-17:
Had this machine just a week or so and it's fantastic, but here's a few things I'd really like implemented in some form:

1. From the **Instrument list,** I'd like the ability to duplicate an instrument
to another slot - with all the settings for it in place. At the moment,
the only way I can see to do this is to load the same sample again
from the SD card into an instrument slot and manually set it up with
the settings again. 

Perhaps holding the 'Instrument' button, scrolling to the instrument
you want to duplicate, pressing copy, and scrolling to the slot you
want to paste it into and then pasting it there could be possible?

I can see other requests for the ability to import/export instruments too - which I'd really like to see.

2. I think it'd be great if you could implement **Note, Instrument,
FX1 and FX2 pop-up windows** to be able to stay open with a quick double click
to open / close, so they stay on the screen, rather than having to
hold the button down each time. 

3. Sample Recorder - is it possible to add a **threshold** setting?

4. I'd also love to see options for **Global Mutes and Pattern Mutes** (like Elektron) - I
think currently they're global mutes for the tracks, but allowing mutes
for patterns would give more flexibility when arranging songs without
feeling the need to be destructive to patterns.

somelamp, 2021-03-19:
you can actually duplicate instruments in a project! I'm not sure if it's documented, but the copy/paste work in the sample loader screen. so it's as easy as highlighting the instrument you want in the list, hitting copy, advancing to a new slot, and pasting

robfenn, 2021-03-19:
> you can actually duplicate instruments in a project! I'm not sure if it's documented, but the copy/paste work in the sample loader screen. so it's as easy as highlighting the instrument you want in the list, hitting copy, advancing to a new slot, and pasting

Ah cool. Thanks for that :) Would be nice if they brought it to the Instruments view while you're working on patterns etc too just for a bit of continuity etc.

closed 2021-12-02 by Jacknametrouble