polyend/TrackerBetaTesting #902: [1.3.1] Exported files often have loud noise.

modmodmodmod, 2021-03-18:
The link below monitors the pattern with protools. This is the noise in two places that are being clipped. This example is v1.3.1, but I have often seen this error in previous versions and other projects.

https://drive.google.com/file/d/1CoY0Q777XUvGNbV2Ldjr6VbTef_HqjjX/view?usp=sharing

bambrose, 2021-03-18:
I've experienced this as well
A blast of white noise when exporting by visible on the waveform. For me it happens at the end of the first bar

clickysteve, 2021-09-26:
[image]

Still an issue in 1.4.0.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS