polyend/TrackerBetaTesting #903: [1.3.1] Freeze when navigate to Volume Destination after setting Granular LFO

hissbackwards, 2021-03-18:
## Freeze when navigate to Volume Destination after setting Granular LFO

### Steps to reproduce

1) In any project (e.g. Create new) and add any sample to Instruments
2) Open Instrument Parameters and open 2nd page
3) Goto Granular Position and select LFO
4) Change step to something fast (e.g. 1/16)
5) Navigate to Volume Destination.

Results:
Freeze.

### Other important information

Firmware: 1.3.1

phacya, 2021-03-19:
Just tried this on my 1.3.1 machione and I confirm this bug too. Hard freeze as you try to move from "Panning" to "Volume".

Zeroapr, 2021-04-15:
Similar problem with me. I adjust the LFO on the Instrument Parameters/"Instrument Automation 2/2"/"Cutoff". Then if I click up the destination it locks up on the Panning. While it's locked up the sound is stuck on the last pitch and can't shut off the Tracker without pulling the plug.

x2db, 2021-04-16:
I've got my unit frozen on this page twice already. Moving up down around Panning and volume. This problem didn't exist in 1.3.0. But I cannot reproduce this 100%.

Zeroapr, 2021-04-16:
Yep. Has happened at least 3 times in my last session. Just seems to happen when you tweak the envelope or LFO in the bottom sections. My pattern isn’t lost but the hard boot is annoying and takes me out of the flow. I hope this gets priority treatment it is a pain in what is otherwise a very solid 1.3.1 firmware.

closed 2021-11-27 by ziej8asek

Jacknametrouble, 2022-01-15:
Fixed in firmware v1.5.0