polyend/TrackerBetaTesting #904: [1.3.1] unable to input notes via midi controller while pattern is playing

somelamp, 2021-03-19:
The steps to reproduce are exactly as expected. While a pattern is playing you can not input notes via a external midi controller while a pattern is playing. The cursor will advance by the set number of steps as if data has been inputted into the channel row, but the step will remain blank. The pads work fine.

However, I swear this use to not be a problem... but I've downgraded to every beta (except b3 cause it's failing for some unknown reason) and last stable (1.3.0), and this bug persists. Probably just crazy, hehe.

Jacknametrouble, 2021-03-19:
Please, try performing the config reset in the Config/Firmware section. Then set the MIDI settings again, and see if it helps.

somelamp, 2021-03-20:
no luck after a config reset </3

if it helps, I'm using the first arturia keystep

Jacknametrouble, 2021-03-20:
Please use https://help.polyend.com/submit_ticket for further troubleshooting.

isbtn01, 2021-03-23:
Hey there, i can confirm this bug, it happens in the following situations:
- if you play a sequence and hit REC+Shift everything is ok and you can add notes normally
- but if you press just REC when the sequence is running, it stops recording notes after the "playback line" goes through your current recording position for the second time (second loop)
cheers!

MortalJS, 2022-02-22:
Hi @isbtn01 @somelamp 
Can you check if the problem occurs in firmware 1.5?

isbtn01, 2022-02-22:
Hey man, just checked it, works fine.
Cheers

closed 2022-02-23 by MortalJS