polyend/TrackerBetaTesting #908: [1.3.1] length of a pattern not respected

loktez, 2021-03-20:
If I copy a pattern to another pattern and change its length on that new pattern, the reading head exceeds the length entered. I have to save, turn off the Tracker and then the length is respected. Also sometimes the pattern displayed does not match the right pattern played.

### Other important information

* How often does this bug happen? Often

* Does this bug happen in previous firmware versions?
Don't think so.

* Does this bug happen in every project?
Yes.

loktez, 2021-03-31:
the video :

https://user-images.githubusercontent.com/76638021/113166990-924e4800-9243-11eb-868b-11a83254ad8e.mov

loktez, 2021-04-06:
It seems that pass to the perform mode and comeback to the pattern mode unlock the good lenght.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS