polyend/TrackerBetaTesting #909: Sample layering utility

twaroshinto, 2021-03-20:
Hi, it would be awesome to have a sample layering utility. I imagine it could be an effect on sample editor. It would allow to load additional sample, sum it up with actually selected sample and render it as a new sample. This way you user would have a great way to create additional sounds in box (like increasing attack by stacking / layering snare with kick drum, chords samples, massive pads etc).
How I imagine it could work: 
- go to sample editor
- select actually loaded sample
- select "Layering" effect
- load additional sample (from card or from samples in memory) - at this point second "waveform" would be displayed under waveform of original sample
- user would tweak parameters like offset (so you could precisely set start of additional sample), cropping/trimming (if you would like to use just a part of loaded sample) and loudness "ratio" of samples, preview result and press save
- at this point new sample would be created in place of old or as a new loaded sample

Thank you for considering this suggestion.
Thanks for all great work you've made already!

anemlife, 2021-03-21:
This would definitely save a lot of time!! +1

notaprogrammer9, 2021-03-22:
Tip:
Have your first (or whichever) pattern as a “sample editor”. In which you tweak, layer and mangle samples directly on the steps. 
If you’re done layering and tweaking, 
SHIFT select your layered samples and bounce into a new sample. 

IMHO this is quicker and more versatile than a layer option in the sample editor.

theorize999, 2021-04-09:
Isn't this the whole reason we have resampling? My humble opinion would be to make the resampling process a little better by having an option to directly resample to an instrument rather than having to export/reload exported sample (it's just a bit messy right now as it makes a new sample on your SD card that will need dealing with later, but I workaround it for now by just making a sample like "Temp Resample 01" and then just overwrite it with the next iteration, but this really should be streamlined in the future), the current functionality should stay in but just some option to resample directly to instrument overwriting any current sample data would speed things up a bit. With resampling you can do better than simple layering up samples, you can filter and apply envelopes to the samples you are layering, it's really way more flexible. This is how I typically layer up drum samples to make a better sample out of two samples, filtering them always makes them sit better with each other for me.

anemlife, 2021-04-09:
@theorize999 you're right about that. 
I was thinking today, what if we could set different locations for renders and recordings? 

I realized that the main issue for my workflow with resampling and recording is that I keep switching between folders when loading samples. (got my own custom sample folder, the export folder for rendered samples and finally the recording folder)

If we could set our own locations it would make things lots faster. (also moving/renaming samples between folders would help.)

But yeah I definitely agree about an option to overwrite an instrument slot when resampling.

theorize999, 2021-04-28:
Yep I agree, it would help a lot. I could organize and do more sound design faster there's a lot of things I haven't even gotten around to trying yet. I imagine the file ops will get better after the bigger issues get taken care of.

closed 2021-12-02 by Jacknametrouble