polyend/TrackerBetaTesting #918: Suggestion - - > Perform mode: filter envelopes + resonance

anemlife, 2021-03-21:
At the moment we can only change the filter LFO speed (in performance mode) and the results are kind of unpredictable and limited. 

What if we got these slots in addition:

- FILTER ENVELOPES: that turns the filter envelopes ON for all selected tracks + we would adjust the multiple (from short to long envelopes = something like ÷10 to x10?). Set the value to OFF to stop the FX. You can adjust the filter type with the Bandpass/Highpass/Lowpass slots

- RESONANCE: to adjust the resonance for all filters.

Two problems I already see with my idea: which FX would take priority if LFO SPEED fx and this ENVELOPPE fx are ON at the same time? Also what about all the other parameters from envelopes? (adsr/amount)

So I'm not 100% sure about my idea but I still think there's something to do with the performance mode and the filter envelopes!

Would love to be able to turn all my samples into sonic/laser sounds with only a few clicks in performance mode!
Any other ideas?

static-noise-PT, 2021-04-20:
Definitely needed would step sound design a lot. At least resonance in Performance Mode would be awesome.

Ixindamix, 2021-04-24:
Hi - Just wanted to add to the resonance in perform mode feature request, it's the only thing I've found missing for my personal performance style.

williamdiffey, 2021-05-17:
+1 for filter resonance as an FX. The filter sweeps aren't really up to par without it