polyend/TrackerBetaTesting #921: MIDI Chords - Intervals and low octave please

emotec, 2021-03-22:
Please add 2 notes intervals to the midi chord list.

01,02,03,04,05,06,07,08,09,0A,0B,0C

Also it would be nice to have a lower octave root note added to all the chords.
So all the chords 037, 047, 048 etc. but with an extra low root octave note!  
These chords could start with 'E' for extended bass.
E37, E47, E48. etc. etc. 

so E47 (Cmaj) would be 'C - C - E - G

thx

modulationsource, 2021-03-23:
That would make the chord feature much more useful.
I already suggested the addition of inversions for the triads, this would only be 4 additional chords,
038, 049, 058 and 059.

emotec, 2021-03-24:
User custom chords would be the ultimate, then we can have whatever we want :-)  That would perhaps be a little difficult to implement though.

jauhoss, 2021-03-24:
yeah, custom chords would be awesome. I like to use often open chords, it would be then possible as well. And also, would be much more intuitive to use if I could pick from the custom chords I have created for the project instead of long list of pre-defined chords.

groovenino, 2021-04-30:
I have to admit that custom chords are a much interesting option. This feature is already requested in  #460 and #579 
Probably in a similar fashion as it is possible already with custom cc midi messages.

closed 2021-12-02 by Jacknametrouble

free-ghz, 2022-04-28:
Agree with this. Coming from other software trackers, this is something i took for granted. Really surprised me that this wasnt the case. It's even using the same notation - like `047` for a major chord - why not just allow arbitrary numbers...? Sometimes you need a 017, or want to do quirky stuf like 0001 for just a little instability. You could do trills and stuff.