polyend/TrackerBetaTesting #923: [1.3.1] Sidechain limiter stops working everytime I process a sample

anemlife, 2021-03-22:
I realised that everytime I process a sample in the sample editor, the sidechain stops working.

I have to move the threshold slightly to make it "catch" the source again.

Also the attack settings get changed, I have to move this one as well to get my settings back.

In this video I show the issue. The sidechain source is my K1 track (big kick) 

https://youtu.be/pgz2gihfTe0

somelamp, 2021-03-28:
It's hard to pin down, but I'm glad ya captured somethin'. I've often had the limiter just stop functioning until I change the threshold or attack </3

anemlife, 2021-03-29:
Yeah this one was pretty sneaky to catch but I'm now 100% sure that it is related to processing samples! (at least every time it happens to me it's because of this reason)

trutypesounds, 2021-09-26:
absolutely the same thing, master limiter sidechain stops working (usually after sample editing as you've mentioned), changing attack nor threshold wont help, reloading project helps.
this still in 1.4.0 @ambv

closed 2021-11-19 by ambv