polyend/TrackerBetaTesting #925: Round Robin Sample Playback

ArtFluids, 2021-03-23:
I would like to suggest a new instrument type/mode.

I want to sample a synth or maybe a drum and do several takes for variation. Then I want the instrument to cycle through those various takes, either at random or in sequence. So let's say I take an 808 high-hat. I don't want to hear the same static sample on every step of my sequence, I want each step to be a slightly different sample for a more natural quality.

Or let's say I want a fun filter sequence from a synth sound like in Dave Wise's Donkey Kong Country music: (at 0:19 here https://youtu.be/RR_ifniLcW4 ). Where each step of the melody is a different sample of a synth with different parameters (like cutoff). It would be nice if the cycling through the samples was automatic. Each step will play the next sample in the group. 

This can already be done with slicing but you would have to manually select the slice for each step. It would be nice if I could just have it be automatic. Cycling through the slices in order, ping pong or at random.

Jacknametrouble, 2021-03-23:
https://www.youtube.com/watch?v=xvlZQ9vY9ck

notaprogrammer9, 2021-03-23:
You can already do this with careful anticipation.
Basically, you have got to “group” your round robin samples together in the sample list. And then (forgot which one) program the correct step FX to cycle through them

closed 2021-03-24 by Jacknametrouble