polyend/TrackerBetaTesting #926: Pattern 49 [1.3.1] Bug title

prds100, 2021-03-23:
I can access pattern 46 with the jog wheel, however no patterns after 49 are lit using the pads like in the first 48 patterns.

If i want to switch to pattern 50 (2nd top left pad), the tracker brings me to pattern 2

Is anyone else experiencing this?  I mentioned before on an older firmware but the issue is still there

clickysteve, 2021-04-02:
I suspect the limitation is the number of pads, right? I'm not sure this is a bug as much it's expected behaviour. How would the Tracker know when you want to access Pattern 50 versus Pattern 2?

pinkcell-niico, 2021-04-02:
Maybe adding Shift key in the shortcut could show and access 49 to 96 Patterns ?
Or a double-click on Pattern screen key (more consistent since it could access 97+ patterns with tripple-click and so on)?

prds100, 2021-04-02:
The number of pads shouldn’t be a limiting factor, if there are more than 48 patterns then the user needs to be able to access patterns after 48. Shift function would be a good idea

MortalJS, 2022-01-13:
Closed. Enhancement

closed 2022-01-13 by MortalJS