polyend/TrackerBetaTesting #928: Share FX Lanes across Tracks

henry-ehlers, 2021-03-24:
## The Problem
I believe to understand that the Polyend Tracker is limited to 2x8 effect tracks owing to its limited RAM, and it is subsequently not possible to simply "add" more effect lanes. Not only that, by the limitation of 2 FX lanes / slots per track is also inherently baked into the tracker's physical button layout. Many however (including me), wish that we could have more FX-modulation per step.

## Suggestion: FX-Lane Sharing 
Here however, I am wondering whether it would be possible for a single instrument to have its effects spread across multiple tracks, i.e. for the Polyend Tracker to introduce effect-lane sharing across tracks. Consider the 3 non-note input types of the note-lane accepts: **OFF** / **CUT** / **FAD**. Here, I wonder whether it would be possible to add a fourth - **EFF**, which would prevent that channel's step's instrument to render any sound, but would allow its two FX-Lanes to apply to any other instance of that instrument on that step.

## An Example
Please consider this crudely rendered example. The following "image" represents the contents of two tracks for a single step.
| Track | Note | Inst. | FX-1 | FX-2 |
| -----------| ----------- | ----------- | ----------- | ----------- |
| TRACK 1 | C5  | 01 | M30 | G25 |                       
| TRACK 2 | **EFF** | 01 | C50  | n10  | 
       
Here, instrument **01** plays note **C5** with two effects from **TRACK 1** applied to it, i.e. micro-tuning of 30 cents, and glide of 25%. In **TRACK 2**, instrument **01** is also present, but does not produce any sound as its note-lane is filled by **EFF**. Instead, the two effects in **TRACK 2** (chance of 50% and random note of +/-10) now apply to **ANY INSTANCE** of instrument **01** playing on that step. If on the next step, **TRACK 2** invoked (say) a hypothetical instrument **02** to play note **A6**, then both those effects (**c50** and **n10**) would immediately stop. This would be in-line with the Tracker's workflow and rules.

## Conclusion
This would effectively give us additional FX-lanes without needing to any more actual channel / lanes, or without destroying the elegant work-flow the Polyend Tracker provides. It would however give us additional modulation power when only few instruments are used per  step. This would be especially for minimal, generative, or ambient work with lots of randomization on a step-by-step basis.

Thank you for considering my suggestion :)

bambrose, 2021-03-24:
That's interesting. It has crossed my mind while using the tracker that I would like the opportunity to sacrifice tracks to enable additional effects. Sort of like the "thru" machine on the elektron octatrack. Your suggestion is in keeping with the workflow I agree

henry-ehlers, 2021-04-04:
To add to the question of what instrument track should be targeted by the additional effects: the simplest conceptual solution (to me at least) may be for all **EFF** track's of a particular instrument to apply to the first *non*-**EFF** instance of that instrument **to its left**. This way there is no ambiguity as to what instrument instance receives additional effects, even if there were multiple sound generating (i.e. *non*-**EFF**) instances of such an instrument. 

This way, in order to check what additional effects apply to a certain instrument on a certain step, all we would need to do is check what instances of this instrument + **EFF** exist to its right on that particular step. The moment another *non*-**EFF** instance of this particular instrument appears, all subsequent **EFF** tracks would apply to it. This strikes me as quite clear and direct to both program and read. 

Just thought I'd mention this here ¯\\_(ツ)_/¯

zach-overflow, 2021-12-26:
I really like this idea! I think your proposal might even be a prerequisite for this enhancement issue I just filed: https://github.com/polyend/TrackerBetaTesting/issues/1339 . I called out this issue in that linked issue as being a likely dependency.