polyend/TrackerBetaTesting #931: Suggestion --> Pattern Editor: no MIDI channel left when deleting a step

anemlife, 2021-03-26:
When you delete a MIDI step, the MIDI channel stays.

I understand that this is supposed to be used when you want to send midi data without notes (since we can send CC, etc...) but I'm only using the PT as a MIDI sequencer so I don't need it. 

Could we get an option in the config menu to disable this behavior so the tracker deletes the full step with MIDI?

0x0zzz, 2021-03-26:
This might be a small regression, but a workaround I found was switching to 'Instrument' mode and then deleting deletes both note and instrument data when the instrument is a MIDI channel.  When in 'Note' mode, the MIDI channel lingers.

anemlife, 2021-03-29:
Yeah that's a good workaround but still annoying as you have to switch between instruments/notes every time.

Let's hope they change this behavior soon :)

tilllt, 2021-04-03:
Yeah, it's annoying to delete the midi instruments manually, also it's not consistent in how the tracker behaves with other (non midi) instruments.